tzapu/SonoffBoilerplate #1: Relay toggles on and off when connected to Arduino IDE

terrag42, 2016-06-02:
I was able to get this sketch working with a WeMos D1 Mini and the first relay of a 4-relay board. The problem I'm having is when I'm connected to and powered by my computer via USB. It's strange, when I plug in the WeMos, it connects to WiFi and behaves normally. Then I open up the Arduino IDE (version 1.6.8) and SOMETHING (not me) starts toggling the relay once per second.
So I thought I'd open up the serial monitor to see if any messages are being transmitted. Surprise! When I open up the serial monitor, it stops toggling and behaves normally. The mysterious toggling behavior correctly interpreted and displayed on my phone by the Blynk app. What is up?

tzapu, 2016-06-02:
now this is weird... i can only think that some default variable/status/ something is not right
opening the serial monitor causes a reset so it means it must work after a reset, which is weird

i ll try to test on a wemos at some point, although, i ve only tried on the sonoff with the serial monitor connected ... might need to test that as well

terrag42, 2016-06-07:
Issue resolved. I was using the Wemos pin "D3" for the button, but that corresponds to GPIO pin 0, which is used for entering programming mode. When the Arduino IDE is running and the WeMos D1 mini board is selected, and without the serial monitor open, it must be causing that pin to toggle about once a second so it's always ready to enter programming mode. So I switched to pin "D8", but that was a bad idea. The way the code is written, the button pin needs to be tied to 3.3v through a ~10k resistor and grounded to trigger the timer to start detecting if it's a long, medium or short press. If WeMos pin D8 (corresponding to GPIO 15) is high when the unit is reset, then it will not run the stored program or enter programming mode. Finally, I assigned pin D5 for button duties. This corresponds to GPIO 14 which doesn't have any special requirements when restarting. 
Note: OTA programming is working! But when I start the Termite terminal program to view serial monitor messages, it resets the ESP. No big deal, but having GPIO 15 (D8) tied high had me confused why the program wouldn't run and it would not enter programming mode.

closed 2016-06-07 by terrag42