tzapu/SonoffBoilerplate #4: boot sonoff into developer mode to flash the firmware

hloun, 2016-07-11:
hello , could you help us how we can boot into developer mode to flash the firmware 
i have tried many ways , i am using FT232RL
[image]
[image]

i connect the sonoff to ft232rl like the picture above ,  then hold the reset button on the sonoff and connect the usb to pc but the  sonoff boot in normal mode so i cant flash anything

tzapu, 2016-07-12:
hi,

do you have per chance a rf enabled sonoff?

cheers

hloun, 2016-07-12:
Yes my sonoff has rf

tzapu, 2016-07-12:
hi,

on the RF i think the button is no longer connected to GPIO 0.

See more here:
http://tech.scargill.net/more-sonoff/

Cheers
Alex

hloun, 2016-07-12:
hi in first special thanks for you 
this way worked with me and i flashed the code 
[image]

but after that  the board going to be useless  because all codes for WIFI VERSION
I think there is something in RF version differents from WIFI version 
thank you

tzapu, 2016-07-12:
hi,

unfortunately, i do not have a rf version to play with.

maybe it s using some gpio wrong, try and remove the code that s gpio dependant...

cheers

kristoffernolgren, 2017-06-23:
@Baha2Odeh what way worked for you, did you short-circuit where the red line is drawn? Did you also hold the button?

Thx