tzapu/SonoffBoilerplate #7: Wemo switch

shenery, 2016-11-14:
Hi, i was wondering if you thing the sonoff switch will work with this code here?

I have installed this firmware on a wemos d1 mini and have since bought some sonoff switches. I wonder if the flash size is too small and also would i have to include any of your libraries or can i just go with the esp8266 ones i have installed?
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/blob/master/README.md

Kind regards
Steven

terrag42, 2016-11-29:
I played around with integrating SonoffBoilerplate and [Xose Perez's Fauxmo library](https://bitbucket.org/xoseperez/fauxmoesp) last night. I could get the Sonoff to respond only if I commented out the virtualWrite command that's inside the setState function:

`void setState(int s) {
  digitalWrite(SONOFF_RELAY, s); 
  digitalWrite(SONOFF_LED, (s + 1) % 2); // led is active low 
//  Blynk.virtualWrite(6, s*255); 
}`
Is there a way to get the LED working?

shenery, 2016-11-29:
If you go to my issue joeman and I have got of working and I'm sure joeman has the led working too but I'm not sure

shenery, 2016-11-29:
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/6#issuecomment-263479950