tzapu/SonoffBoilerplate #8: 'MQTT' does not name a type

sticks221, 2017-02-06:
Hi,

Thank you for sharing this project - it looks to be exactly what I need however,  when trying to compile, I get "exit status 1" and the above error with line 299 highlighted.  Please firgive me if this is a silly question but I am new to this - I have double checked that the dependancies are present (esp8266:v2.3.0, BLYNK:v.04.5 and WiFiManager:v0.12.0) but I just cant work it out.

Please can you advise?

tzapu, 2017-02-06:
hmm, maybe it s a different pub sub library?

i am using the form made by lmroy, and the version seems to be 1.99.1.

good luck and thanks

sticks221, 2017-02-06:
Thanks for the prompt reply,  would you be able you share the library you are using (or a link to it) please?  Google keeps taking me back to the same library that I have installed already.  I have downgraded to v2.2.0 but still no joy.

tzapu, 2017-02-06:
https://github.com/Imroy/pubsubclient <https://github.com/Imroy/pubsubclient>

try this

cheers

sticks221, 2017-02-08:
Brilliant - thank you,  thats the one I needed.  I did have to delete the existing one from the IDE first though (My Documents\Arduino\libraries\PubSibClient) in windows.  It seems that this library is newer than the one in the IDE but causes a clash if trying to install on its own.

closed 2017-02-08 by sticks221