tzapu/SonoffBoilerplate #11: Error during compiling with MQTT support

katodo, 2017-04-30:
Without MQTT support enabled build procedure work well but if I install PubSubClient V 2.6 and enable MQTT support I've this error, can you help me?
 

`SonoffBoilerplate.ino:299: error: 'MQTT' does not name a type
 void mqttCallback(const MQTT::Publish& pub) {

SonoffBoilerplate.ino:299: error: expected unqualified-id before '&' token
 void mqttCallback(const MQTT::Publish& pub) {

SonoffBoilerplate.ino:299: error: expected ')' before '&' token
SonoffBoilerplate.ino:299: error: expected initializer before 'pub'
 void mqttCallback(const MQTT::Publish& pub) {

 D:\Dropbox\Project_Arduino\SonoffBoilerplate_original\SonoffBoilerplate.ino\SonoffBoilerplate.ino.ino: In function 'void updateMQTT(int)':

SonoffBoilerplate.ino:161: error: no matching function for call to 'PubSubClient::publish(char [50], String&)'
   mqttClient.publish(topic, stateString);`

tzapu, 2017-05-01:
hi,

might be the wrong pbsub try the one annotated in code
https://github.com/Imroy/pubsubclient <https://github.com/Imroy/pubsubclient>

cheers

katodo, 2017-05-01:
Ok, tnx! I've solved using suggested lib.
During my first test I've used library available from Arduino Lib Manager... but it's a different lib with the same name :)

closed 2017-05-01 by katodo