tzapu/SonoffBoilerplate #12: Error compiling SonoffBoilerplate.ino

iagson, 2017-05-03:
The following error appears everytime I try to compile your .ino. 
Is there anything I might be missing?

Thanks in advance!

> Arduino:1.8.1 (Windows 7), Tarjeta:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"
> 
> SonoffBoilerplate:299: error: 'MQTT' does not name a type
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                          ^
> 
> SonoffBoilerplate:299: error: expected unqualified-id before '&' token
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                                       ^
> 
> SonoffBoilerplate:299: error: expected ')' before '&' token
> 
> SonoffBoilerplate:299: error: expected initializer before 'pub'
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                                         ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\SonoffBoilerplate-master\SonoffBoilerplate\SonoffBoilerplate.ino: In function 'void updateMQTT(int)':
> 
> SonoffBoilerplate:161: error: no matching function for call to 'PubSubClient::publish(char [50], String&)'
> 
>    mqttClient.publish(topic, stateString);
> 
>                                         ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\SonoffBoilerplate-master\SonoffBoilerplate\SonoffBoilerplate.ino:161:40: note: candidates are:
> 
> In file included from D:\Programas\arduino-1.8.1\SketchFolder\SonoffBoilerplate-master\SonoffBoilerplate\SonoffBoilerplate.ino:49:0:
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:130:12: note: boolean PubSubClient::publish(const char*, const char*)
> 
>     boolean publish(const char* topic, const char* payload);
> 
>             ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:130:12: note:   no known conversion for argument 2 from 'String' to 'const char*'
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:131:12: note: boolean PubSubClient::publish(const char*, const char*, boolean)
> 
>     boolean publish(const char* topic, const char* payload, boolean retained);
> 
>             ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:131:12: note:   candidate expects 3 arguments, 2 provided
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:132:12: note: boolean PubSubClient::publish(const char*, const uint8_t*, unsigned int)
> 
>     boolean publish(const char* topic, const uint8_t * payload, unsigned int plength);
> 
>             ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:132:12: note:   candidate expects 3 arguments, 2 provided
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:133:12: note: boolean PubSubClient::publish(const char*, const uint8_t*, unsigned int, boolean)
> 
>     boolean publish(const char* topic, const uint8_t * payload, unsigned int plength, boolean retained);
> 
>             ^
> 
> D:\Programas\arduino-1.8.1\SketchFolder\libraries\PubSubClient\src/PubSubClient.h:133:12: note:   candidate expects 4 arguments, 2 provided
> 
> D:\Programas\arduino-1.8.1\SketchFolder\SonoffBoilerplate-master\SonoffBoilerplate\SonoffBoilerplate.ino: At global scope:
> 
> SonoffBoilerplate:299: error: 'MQTT' does not name a type
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                          ^
> 
> SonoffBoilerplate:299: error: expected unqualified-id before '&' token
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                                       ^
> 
> SonoffBoilerplate:299: error: expected ')' before '&' token
> 
> SonoffBoilerplate:299: error: expected initializer before 'pub'
> 
>  void mqttCallback(const MQTT::Publish& pub) {
> 
>                                         ^
> 
> exit status 1
> 'MQTT' does not name a type
> 
> Este reporte podría tener más información con
> "Mostrar salida detallada durante la compilación"
> opción habilitada en Archivo -> Preferencias.
>

iljkaersuoasgr, 2017-06-05:
I am experiencing a very similar litany of errors. I'm using arduino 1.8.2 with wifimanager, blynk, and pubsubclient installed via the built in library manager. Exact messages pasted below.

`Arduino: 1.8.2 (Windows 8.1), Board: "Generic ESP8285 Module, 80 MHz, 115200, 1M (512K SPIFFS)"

Build options changed, rebuilding all
SonoffBoilerplate:299: error: 'MQTT' does not name a type

 void mqttCallback(const MQTT::Publish& pub) {

                         ^

SonoffBoilerplate:299: error: expected unqualified-id before '&' token

 void mqttCallback(const MQTT::Publish& pub) {

                                      ^

SonoffBoilerplate:299: error: expected ')' before '&' token

SonoffBoilerplate:299: error: expected initializer before 'pub'

 void mqttCallback(const MQTT::Publish& pub) {

                                        ^

C:\Users\Ben\Documents\Project Files\Internet of Bens Things\SonoffBoilerplate\SonoffBoilerplate\SonoffBoilerplate.ino: In function 'void updateMQTT(int)':

SonoffBoilerplate:161: error: no matching function for call to 'PubSubClient::publish(char [50], String&)'

   mqttClient.publish(topic, stateString);

                                        ^

C:\Users\Ben\Documents\Project Files\Internet of Bens Things\SonoffBoilerplate\SonoffBoilerplate\SonoffBoilerplate.ino:161:40: note: candidates are:

In file included from C:\Users\Ben\Documents\Project Files\Internet of Bens Things\SonoffBoilerplate\SonoffBoilerplate\SonoffBoilerplate.ino:49:0:

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:130:12: note: boolean PubSubClient::publish(const char*, const char*)

    boolean publish(const char* topic, const char* payload);

            ^

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:130:12: note:   no known conversion for argument 2 from 'String' to 'const char*'

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:131:12: note: boolean PubSubClient::publish(const char*, const char*, boolean)

    boolean publish(const char* topic, const char* payload, boolean retained);

            ^

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:131:12: note:   candidate expects 3 arguments, 2 provided

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:132:12: note: boolean PubSubClient::publish(const char*, const uint8_t*, unsigned int)

    boolean publish(const char* topic, const uint8_t * payload, unsigned int plength);

            ^

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:132:12: note:   candidate expects 3 arguments, 2 provided

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:133:12: note: boolean PubSubClient::publish(const char*, const uint8_t*, unsigned int, boolean)

    boolean publish(const char* topic, const uint8_t * payload, unsigned int plength, boolean retained);

            ^

C:\Users\Ben\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:133:12: note:   candidate expects 4 arguments, 2 provided

C:\Users\Ben\Documents\Project Files\Internet of Bens Things\SonoffBoilerplate\SonoffBoilerplate\SonoffBoilerplate.ino: At global scope:

SonoffBoilerplate:299: error: 'MQTT' does not name a type

 void mqttCallback(const MQTT::Publish& pub) {

                         ^

SonoffBoilerplate:299: error: expected unqualified-id before '&' token

 void mqttCallback(const MQTT::Publish& pub) {

                                      ^

SonoffBoilerplate:299: error: expected ')' before '&' token

SonoffBoilerplate:299: error: expected initializer before 'pub'

 void mqttCallback(const MQTT::Publish& pub) {

                                        ^

exit status 1
'MQTT' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

tzapu, 2017-06-05:
hi,

you need the pubsub lib by lmroy i think, it s not in the library manager

cheers

iljkaersuoasgr, 2017-06-06:
This worked. I replaced the files in "Documents/Arduino/Libraries/Pubsubclient" with the files from https://github.com/Imroy/pubsubclient an it compiled perfectly. Thanks!